owntech-foundation/Core #2: Error even after a first flash with bootloader

twalter23, 2024-02-15:
Flashing code with bootloader without pressing boot and reset button is not possible, got the following error:

`AVAILABLE: blackmagic, cmsis-dap, custom, jlink, mbed, stlink
CURRENT: upload_protocol = custom
upload_pre(["upload"], [".pio\build\bootloader_serial\firmware.mcuboot.bin"])
Detecting Spin boards connected to host...
  Board 1: Found Spin board on port COM38 with unique ID 4630500400360057
Uploading to board with ID 4630500400360057
Uploading .pio\build\bootloader_serial\firmware.mcuboot.bin
'MCUMGRPATH' n'est pas reconnu en tant que commande interne
ou externe, un programme ex\x82cutable ou un fichier de commandes.
*** [upload] Error 1`

twalter23, 2024-02-15:
flashing after pressing boot and reset button works

luizvilla, 2024-02-16:
@twalter23 : Can you please be more precise in your issue?
Which code did you use?

luizvilla, 2024-02-16:
@twalter23 I have tried flashing a spin board using the core. Here's my procedure: 
- Checked out the main branch of the core
- Flashed the bootloader onto the spin board
- Compiled and flashed the core
- My upload speed was 14kB/s
- The LED was blinking as expected

twalter23, 2024-02-16:
I used this code:
https://github.com/twalter23/githubCore

jalinei, 2024-02-28:
Root cause of this issue was found : 
Wrong version of the bootloader@V1.0 was deployed on board instead of bootloader@V1.1
I'm closing.

closed 2024-02-28 by jalinei