owntech-foundation/Core #5: [mcumgr] Automatic download for mcumgr in MAC not workingTWIST_v_1_1_2

luizvilla, 2024-02-26:
**Description**

While working with the Tutorials with some students, I had one of them with a MAC computer that was not able to flash the twist board. 

**Fix**

We had to manually download  `https://github.com/owntech-foundation/mcumgr/releases/download/v0.4/mcumgr-mac`
and place it in `owntech/third_parties`. 

**To do**

Try to reproduce this problem on another MAC and figure out why it did not work.

jalinei, 2024-02-28:
@guigur Have you experienced a similar issue ? We need more feedback in order to understand if it is a good first issue.

luizvilla, 2024-04-16:
In the meantime a quick fix : 
- mcumgr can be downloaded here : https://github.com/owntech-foundation/mcumgr/releases/tag/v0.4
- take the mac version and place it in your project in the repo owntech/third_party/mcumgr-mac
- The bug root cause should be in the automatic downloading script is in owntech/scripts/pre_bootloader_serial.py
- Here is the source code that does not work:
  ```py 
  
  Import("env")
  
  import subprocess
  import serial.tools.list_ports
  import os
  import platform
  import helper_functions
  import platformio
  from platformio.device.list.util import list_serial_ports
  from time import sleep
  
  ################### Make sure Mcumgr exists locally ###################
  
  # Determine executable name
  third_party_dir = os.path.join(".", "owntech", "third_party")
  mcumgr_executable = "mcumgr"     if platform.system() == 'Linux' and platform.machine() == 'x86_64' else \
                      "mcumgr-rpi" if platform.system() == 'Linux' and platform.machine() == 'armv7l' else \
                      "mcumgr-mac" if platform.system() == 'Darwin'  else \
                      "mcumgr.exe" if platform.system() == 'Windows' else \
                      None
  mcumgr_url = "https://github.com/owntech-foundation/mcumgr/releases/download/v0.4/" + mcumgr_executable
  
  res = helper_functions.check_file_and_download("Mcumgr", third_party_dir, mcumgr_executable, mcumgr_url, True)
  
  if res != 0:
      import warnings
      warnings.warn("Unable to download Mcumgr! Make sure you are connected to the Internet.", Warning)
      warnings.warn("Without this tool, you may not be able to upload the firmware to the board without a ST Link.", Warning)
  
  mcumgr_path = os.path.join(third_party_dir, mcumgr_executable)
  ```

jalinei, 2024-04-19:
Reported not to work on Macbook Pro M-1 (13-INCH 2020) latest software Sonoma 14.4.1. 
bcabc171ae2af575af11e6f522f35e10773d336f has not solved the issue. 
@guigur can you give a deeper look ?

guigur, 2024-04-19:
The problem does not seem to be code-related. Waiting for response of anne

guigur, 2024-05-21:
The problem lies in the `helper_function.check_file_and_download()`.
The function's return is not 0; therefore, we get the error `Unable to download Mcumgr! Make sure you are connected to the Internet`

The student had internet and tried compiling at home with a different network.
We allowed the terminal.app aswell as vscode.app to access the whole disk.
Then we did allow all the rights to the `third_party`  folder with a `chmod 777 third_party` with no effects.
The helper function has created the folder for the binary but did not download the `mcumgr` executable.

The culprit lies in the file helper_functions.py
```
        if not os.path.isfile(file_path):
		import urllib.request
		from urllib.error import URLError

		print(f"{what} is not available locally, downloading.")
		try:
			# Download file
			urllib.request.urlretrieve(file_url, file_path)
			print("Download complete.")
		except URLError:
			print(f"Error! Unable to download {what}.")
			print("Make sure you are connected to the Internet.")
			return -1
```

jalinei, 2024-05-21:
So it is because the URL is wrong right ?

guigur, 2024-05-22:
No, I made him print the URL and afterward, we copy-pasted the URL with a `wget` command, and the correct file was downloaded. I don't know what's going on.

It's either a mac thing **again** or some voodoo magic. 

For now, the student is no longer stuck and can use the code by manually downloading the file.

As for myself, I'll wipe my old MacBook to reinstall from scratch the core.

guigur, 2024-06-17:
okay, I have encountered the issue by myself. The problem is related to SSL certificates. I'll dig deeper
``` C
URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)>:
```

guigur, 2024-07-25:
I found a workaround. needs to be tested on other systems

closed 2024-10-10 by luizvilla

adrienprevost, 2025-07-22:
Hi, I have another issue with mcumgr on Windows 11. Each time I want to upload some code to spin, during the upload process, platformIO looks for mcumgr `Mcumgr is not available locally, downloading.`. If my internet connexion is through the lab network, my proxy would block the downloading of mcumgr. I bypassed the issue by changing the internet connexion channel and deactivate the proxy. However, I wonder why mcumgr needs to be downloaded each time.
Shall we open another issue about this, or since this is about mcumgr, we keep on the same thread ?

PR #80 (2024-10-10): fix macos mcumgr in  helper_functions.py
fix #5 
this replaces the legacy `urlretrieve` for downloading external resources
This needs to be tested on Win and Linux
commit 104f877e (2024-10-10): fix macos mcumgr in  helper_functions.py (#80)

fix #5 
this replaces the legacy `urlretrieve` for downloading external
resources
This needs to be tested on Win and Linux