owntech-foundation/Core #10: Cmakelist of Communication API is not up to date

luizvilla, 2024-03-04:
**Description**

While trying to compile the core we came across an error pointing to the CMakeList of the communication API. 

[image]

**Solution**

The file names in the CMakeList are outdated and still have `xxxCommunication` attached to them. This has to be suppressed.  

```c
if(CONFIG_OWNTECH_COMMUNICATION)
  # Select directory to add to the include path
  zephyr_include_directories(./public_api)

  # Define the current folder as a Zephyr library
  zephyr_library()
  
  zephyr_library_sources(
    public_api/CommunicationAPI.cpp
  )

  if (CONFIG_OWNTECH_COMMUNICATION_ENABLE_ANALOG)
    zephyr_library_sources(
      src/AnalogCommunication.cpp
    )
  endif()

  if (CONFIG_OWNTECH_COMMUNICATION_ENABLE_CAN)
    zephyr_library_sources(
      src/data_objects.cpp
      src/thingset_isotp.cpp
      src/thingset_pub_sub.cpp
      src/CanCommunication.cpp
    )
  endif()

  if (CONFIG_OWNTECH_COMMUNICATION_ENABLE_RS485)
    zephyr_library_sources(
      src/Rs485Communication.cpp
      src/Rs485.cpp
    )
  endif()

  if (CONFIG_OWNTECH_COMMUNICATION_ENABLE_SYNC)
    zephyr_library_sources(
      src/SyncCommunication.cpp
    )
  endif()

endif()
```

Ayoub-Farah, 2024-03-04:
The file names present in the CMake are correct. Here is what we have in the communication API folder : 

```
├── CMakeLists.txt
├── Kconfig
├── module.yml
├── public_api
│   ├── CommunicationAPI.cpp
│   └── CommunicationAPI.h
└── src
    ├── AnalogCommunication.cpp
    ├── AnalogCommunication.h
    ├── CanCommunication.cpp
    ├── CanCommunication.h
    ├── data_objects.cpp
    ├── data_objects.h
    ├── Rs485Communication.cpp
    ├── Rs485Communication.h
    ├── Rs485.cpp
    ├── Rs485.h
    ├── SyncCommunication.cpp
    ├── SyncCommunication.h
    ├── thingset_isotp.cpp
    └── thingset_pub_sub.cpp
    ```
I can't reproduce the issue, I don't have any error concerning CMake.

martinjaeger, 2024-03-04:
Originally, this error happened when I enabled `CONFIG_OWNTECH_COMMUNICATION_ENABLE_CAN=y` in `zephyr/prj.conf` and commented out the following lines in `zephyr/modules/owntech_communication/zephyr/CMakeLists.txt` in order to avoid the compile errors because of outdated ThingSet library:

```
  if (CONFIG_OWNTECH_COMMUNICATION_ENABLE_CAN)
    zephyr_library_sources(
      src/data_objects.cpp
      src/thingset_isotp.cpp
      src/thingset_pub_sub.cpp
      src/CanCommunication.cpp
    )
  endif()
```

For some reason, this is working now. I think this issue can be closed. I will open a new one for the ThingSet module updates where I can't figure out how to make the build system pick up the Kconfig of the modules.

closed 2024-03-05 by Ayoub-Farah