owntech-foundation/Core #11: Possibility to see hidden folders in VS code

martinjaeger, 2024-03-04:
Is there a possibility to "un-hide" the folders that don't show up in VS code other than deleting the file `.vscode/settings.json`?

Deleting the file is not ideal, as this file is tracked by git and I will always have to take care that I don't accidentally check it in.

Ayoub-Farah, 2024-03-05:
To see the hidden folder, you need to activate the expert mode by following these steps : 
 
 1. Click on platformIO icon in the left side of Vscode
 2. Within the project tasks, select the USB folder
 3. You will find the option "toggle expert mode" in the Owntech folder
 
 
[image]

martinjaeger, 2024-03-06:
Thanks, didn't spot that menu entry.

However, this has a similar effect as deleting the file. Now I still have to take care not to check in the updated `.vscode/settings.json` accidentally.

IMO this file should not be checked in and should be added to `.gitignore`.

jalinei, 2024-04-19:
Quickfix `git update-index --skip-worktree`.vscode/settings.json`

We need to find a way to upstream that, but there is no obvious way from what I understand. We do need to upstream settings.json to hide files in non expert context. 
Alternative would be to build a PIO script to add desired configuration to settings.json dinamically and remove settings.json from upstream.

martinjaeger, 2024-04-29:
Ok, no worries. The command is fine for me.

Thanks for following up, @jalinei and @cfoucher-laas!

martinjaeger, 2024-05-15:
I found that the following command is even better because it seems to ignore the changed file even when pulling in new changes:

```
git update-index --assume-unchanged .vscode/settings.json
```