owntech-foundation/Core #13: [DataAPI] Improve redeability by adding explicit ADC definitions

luizvilla, 2024-03-06:
**Description**

When using the DataAPI, there are function calls which require explicitly naming an ADC. This should be improved for better redeability. 

**Solution**
Two solutions are possible : 
- Add a simple `#define ADCx x` where x is a natural number from 1 to 5 (since there are 5 ADCs to the STM32G474RE of the spin board)
Example: 
```c
#define ADC1 1
#define ADC2 2
#define ADC3 3
#define ADC4 4
#define ADC5 5
```
- Create a typedef enum which lists these ADCs by hand 
```c
typedef enum
{
	ADC1 = 1,
	ADC2 = 2,
	ADC3 = 3,
	ADC4 = 4,
	ADC5 = 5
} ADC_t;
```
- Create a typdef enum which automatically parses the adcs present in the device tree 

```c
#define ADC_TOKEN(node_id) DT_STRING_TOKEN(node_id, adc_number),

typedef enum
{
	UNDEFINED_ADC = 0,
	DT_FOREACH_STATUS_OKAY(st,stm32-adc, ADC_TOKEN)
} ADC_t;
```

cfoucher-laas, 2024-03-12:
Hello @luizvilla,

At the time I implemented it, we quickly discussed this and agreed that just providing a number instead of an enum was easier for newcomers that may not be aware of the provided enumerations.
I guess this is mainly a matter of correctly documenting it. Anyway, since that discussion, we already introduced a bunch of enums (e.g. for shield channels), thus we can reconsider this initial choice and move toward an enum.

However, ADC1, ADC2, etc. symbols already exist from the LL layer, thus we can't redefine them. We could use these sylols as parameters, however I would recommend against it, as this would introduce low-level definitions in a user-level API, and add weird types in functions signatures.

Have you though of another meaningful names we could use?

jalinei, 2024-04-19:
To move forward on this : 
User is expected to refer to the Pinout image that contains required info for ADC configuration 
https://owntech-foundation.github.io/Documentation/core/docs/adc/#__tabbed_1_1

`ADCx` make sense and is optimal for the user experience. Even if it is already defined by LL layer, we should find a way of reusing it there..

jalinei, 2024-04-19:
Increasing the priority level of that issue, we have to solve all discrepancies between ADC HAL and Data API to move forward with the ongoing documentation process

cfoucher-laas, 2024-04-29:
I begin working on this issue today.

If you really want to use ADCx, I think the proper way would be to to rename the existing type using a `typedef`  in order to avoid including low level types names in the user-level API, but this is not very clean practice.

cfoucher-laas, 2025-06-11:
Adressed in v1.0