owntech-foundation/Core #25: [PowerAPI] Overhaul of the PowerAPI to enable automatic LEG object creation

luizvilla, 2024-04-18:
## Description

The function calls of the PowerAPI are written on a way that hampers their understanding and usage. 
For example: 

[image]

Notice how several of the function calls are redundant and attached to a LEG type object. 

It is only necessary to use: 

- `init`: to initialize a leg on with a certain conversion (`BUCK` or `BOOST`) and with a certain mode (`VOLTAGE` or `CURRENT`)
- `set`: to set a certain parameter of the `LEG` (`AdcDecim`, `DutyCycle`, `DeadTime`, `PhaseShift`, `SlopeCompensation`, etc...) 
- `start`: to start a certain LEG
- `stop`: to stop a certain LEG

## Action

Explore a way to automatically create objects within the twist class that setup the parameters of the LEG and exposes its functions.

For instance, this is used in the `TwistAPI.h` already to initialize the `LEG` index: 

```c

#define LEG_TOKEN(node_id) DT_STRING_TOKEN(node_id, leg_name),

/* All the leg with okay status in the device tree */
typedef enum
{
	DT_FOREACH_CHILD_STATUS_OKAY(DT_NODELABEL(power_shield), LEG_TOKEN)
} leg_t;

```

This automatic creation should be driven by the `dts` file that holds all the information of the LEGs.  For instance, the `pwm_legs.dtsi` has the following code: 

```d

leg1: leg1{
    leg-name = "LEG1";
    pwm-pin-num = <12 14>;
    current-pin-num = <30>;
    adc-trigger = "ADCTRIG_1";
    default-adc-decim = <1>;
    edge-trigger = "EdgeTrigger_up";
    default-dead-time = <100 100>;
    modulation = "UpDwn";
    default-phase-shift = <0>;
};

```

jalinei, 2024-04-19:
I don't understand the scope of the proposed modification. Is it modifying Twist API only or multiple APIs ? 
If Twist API only, please edit your issue so the tag is [Twist API] and do not refer as PowerAPI as the scope is to broad.

luizvilla, 2024-04-19:
All power boards APIs should be created automatically from the dts. 

The PowerAPI should implement this automation.

jalinei, 2024-07-08:
@luizvilla I'm not sure to understand this issue. Can you elaborate ? 
At the moment the powershield dts fragment to define the default behavior of the power leg.

luizvilla, 2024-07-08:
This was written a long time ago, before you started writing on the shield API.
The idea is to use the dts to automatically set the power api, which is something I think you are sort of working on.
Thus means having a similar behavior between the initialization of the dataAPI and the power part of the shield api.

jalinei, 2025-01-27:
This has been implemented in v1.0.0

closed 2025-01-27 by jalinei