owntech-foundation/Core #27: [DataAPI] Having a function to retrieve a vector of **converted** values

jalinei, 2024-04-19:
At the moment we have two options: 

- Retrieve the last converted value 
- Retrieve an array of raw values 

Retrieving an array of converted values should be added.

cfoucher-laas, 2024-05-22:
Hello @jalinei,
This one has a some tricky bits in it, even if it seems easy at first glance.

The main issue is that, in C++, arrays are not dynamic. On the programming side, the easiest and cleanest way to do would be to allocate an array on-the-fly (using `malloc`), and to rely on the user to de-allocate it when it is done with it (using `free`). However, this would be non-user friendly for the intended audience..

The second way to do, as is done with the raw values, would be to statically assign two arrays, and rotate them in background when the function is called. This, however, would be very memory-consuming, as to do so, we require two arrays of float_32 (4 bytes) for each enabled channel. To determine the size of the arrays, we may be able to mimic the size of the raw values, I still have to check that.
This memory consumption would be permanent, as soon as the API is started, whether the feature is used or not.

A better solution would be to introduce modern C++ features such as `shared_ptr`, but this would be very non-user friendly and cumbersome. Moreover, I'm pretty sure Zephyr does not support this subset of C++.

Any thoughts about it?

jalinei, 2024-05-22:
Maybe a convenient middle-ground would be to fix a maximal size of the arrays ? 

Let say that the user call a function: data.getConvertedValues(I_Low, 10) 
He would retrieve the last 10 values, converted. 

This way the user has leverage on the memory footprint taken by the dataacquisition internals ? 
We could also hard code a max number of values so that in case we go crazy on measurement numbers we do not overflow ?

Let me know if it is even doable, I guess it might be hard to have arrays that are not matching rawvalue size.

jalinei, 2024-05-22:
Can you give me a relative metric of the memory footprint increase ? From what I understand raw values are stored with a uint16_t format, so adding converted arrays to the API would increase the memory footprint by a factor 2 ? 

What is the current memory footprint ? Is that problem dramatic ? Is it dramatic in some edge case only ? e.g slow control task & many measurements ? 

I understand the issue would be calling a control task based on timer6 and having a fast PWM triggering conversion leading to huge buffer size. 
How do we deal with this edge case in the current raw value implementation ?

cfoucher-laas, 2024-05-22:
`Let say that the user call a function: data.getConvertedValues(I_Low, 10)
He would retrieve the last 10 values, converted.`

Yes, I had this in mind too, however, this would require an additional output parameter to indicate the number of values effectively acquired, as with `getRawValues`, which may be confusing to the user as to which means what.

`From what I understand raw values are stored with a uint16_t format, so adding converted arrays to the API would increase the memory footprint by a factor 2 ?`
Actually, the footprint would be x3, as we would still have the current and unavoidable raw values arrays.

E.g. for a classic Twist acquisition (6 sensors), if we want 10 acquisitions on each sensor between two reads, we would have: 
- 10 acquisitions x 6 sensors x 2 bytes x 2 arrays = 240 bytes for the raw values alone,
- compared to 240 x 3 = 720 bytes if we want to be able to directly read the converted values.
This is an overhead of more than half a kibibyte.

cfoucher-laas, 2024-05-23:
Hello,
_after having slept on it_
actually, I just thought that the mechanism I use for raw values (double-buffering, static buffer allocation) is not required for converted values, as values are converted on-the-fly on function call.
The fact that the double-buffering is not required divides by 2 the memory footprint. Computations from the above message become:
- 10 acquisitions x 6 sensors x 2 bytes x 2 arrays = 240 bytes for the raw values alone,
- compared to 240 x 2 = 480 bytes if we want to be able to directly read the converted values.
Moreover, as double-buffering is not required, we can also allocate the memory on the first call of the function in order to not use the memory if the function is never called. However, this would add a time overhead on the first call to allow for the allocation of the buffer.

I think this is the path I will follow, as this function will already be a big time-glutton, adding an additional overhead is lesser a pain than having a portion of the memory that is immobilized and potentially unused. We can even envision a KConfig switch to change the default behavior.

jalinei, 2024-05-23:
What is the current behavior if the control task frequency is not 10kHz ? 
Would we still obtain 10 values per channel or more ? 

Is the buffer size related to the controlTask frequency and triggering frequency ?

cfoucher-laas, 2024-05-24:
When the dispatch is done by a control task, the number of values stored in the buffers depends on the frequency of the PWM, of the task, and on the number of enabled DMA channels.
By default, one acquisition on one DMA channel is done at each PWM cycle, and a dispatch is done at each task period. This gives a number of measurement, which is rounded up to a multiple of the channels count to ensure measurement are always at the same position in the buffer. There are two buffers per ADC. This part gives the size of the ADC/DMA buffers.

Regarding the size of the arrays that are provided to the user (after dispatch), I see that I did chose a fixed value at the time (32 slots per pin/sensor x 2 arrays per pin/sensor due to double-buffering), which is an additional memory footprint. I do not remember how this value was chosen, and I think we can recalibrate it. Thing is, if I recall the discussion about it, that we can not assume that the application will read value (and empty user buffers) at each task cycle, thus we provided a bit more. We could however define this value, either as a Kconfig value, or use a Kconfig switch to chose between a fixed number or a dynamic value, I think.