owntech-foundation/Core #29: [dataAPI] Missing generic function to store calibration parameters for SPIN.

jalinei, 2024-04-19:
We need the counterpart of `data.setTwistChannelsUserCalibrationFactors()` for a generic data acquisition using SPIN.

luizvilla, 2024-04-19:
I have created a function to do this. You can find it here : 

https://gitlab.laas.fr/lflavado/core/-/tree/power_test_boot?ref_type=heads

In the `DataAPI.h` you will find: 

```cpp

	/**
	 * @brief Use this function to write the gain and offset parameters of the board to is non-volatile memory.
	 *
	 * @note  This function should be called after updating the parameters using setParameters.
	 *
	 * @param channel Name of the shield channel to save the values.
	 */
	int8_t saveParametersInNVS(channel_t channel);

	/**
	 * @brief Use this function to read the gain and offset parameters of the board to is non-volatile memory.
	 *
	 * @param channel Name of the shield channel to save the values.
	 */
	int8_t getParametersFromNVS(channel_t channel);

```

This is only implemented for the TWIST board, but it is not difficult to generalize from here if needed.

jalinei, 2024-04-19:
I propose the following : 
1. Draft a PR based on these two commits. From what I understand It it already complementary to existing data.setTwistChannelsUserCalibrationFactors() which is fully manual and based on console interaction. (Which is not suitable in many contexts) @luizvilla 
2. Let's discuss how to adapt is to a generic form for SPIN. @luizvilla @jalinei 
3. Merge.
4. Add section to documentation @jalinei

luizvilla, 2024-04-19:
Sounds good to me. I can prepare a PR.

jalinei, 2024-04-29:
#34 Is closing this issue right ?

closed 2024-07-09 by jalinei