owntech-foundation/Core #30: Adding an issue template

jalinei, 2024-04-19:
Can someone draft a template for future issues ?

luizvilla, 2024-04-19:
I propose: 

## Context

Briefly explain how you came across this issue. Answer questions such as: 
- What were you trying to do? 
- Which hardware or software version are you working with?
- How did you realize this was an issue?
- Is this specific to a case or a more generic issue? 
- 

## Description

Briefly describe the issue itself:
- Explain the code you tried to write
- If it was working before, reference the code that is sure to work through a commit SHA for instance
- If you have any error messages, please paste them here
- If you tried to understand the error, paste any relevant links

## Solution

Briefly describe the solution for the issue, if you have any: 
- If you have a clear idea on how to solve the issue, please describe it
- If you do not know how to solve the issue, please describe the expected result

Do not hesitate to create a pull request if you think you can solve the issue. But please create the issue first.

jalinei, 2024-05-02:
Done by @luizvilla today.

closed 2024-05-02 by jalinei