owntech-foundation/Core #35: [RFC] Reorganize user-level Data API

cfoucher-laas, 2024-04-30:
This issue is a [RFC](https://en.wikipedia.org/wiki/Request_for_Comments)-like discussion aiming at proposing a new organization for the user-level Data API.

### Motivation
- There are various issues open regarding Data API and ADC HAL such as #29, #28 or #13.
- There have been an ongoing internal discussion about the ADC HAL (currently provided as a `spin.adc` object to the end-user) having been hollowed-out by changes implemented last year, where its internal functions where dispatched into the lower-layer ADC driver and the higher-layer Data API (currently provided as a `data` object to the end-user). Few useful functions remains in the `spin.adc` HAL.
- There have been an internal discussion about providing to the end-user API entry-points that are related to the actual hardware for easier function lookup. This could be carried out by splitting the `data` object functions into a `spin.data` and a `shield.sensor` entry points.
- The current documentation organization highlighted that there is a very high overlap between ADC HAL and Data API, to the point where both pages introduces functions calls from the other API.

The current separation between `spin.adc` and `data` being clearly obsolete, any work on these APIs code or documentation will have to be reworked again in a short time when we reconsider the separation. The current time is a good time to revisit the separation between these APIs.

### Proposal

The proposal is as follows:
- Remove the `spin.adc` entry point that have low added-value to the user.
- Split the `data` entry point under the names `spin.data` and `shield.sensors`. The new `spin.data` entry point could host the two remaining useful functions from `spin.adc` that currently have no equivalent into `data`: `configureTriggerSource()` and `configureDiscontinuousMode()`.
- As part of this change, all references to `channels` for shields would be renamed as `sensors` to avoid confusion with ADC channels, which are a different notion.

The new `shield` object can be the starting point for automatically-adjusting code (using the device tree), as currently done by all functions in `data` that are related to the Twist shield. This code currently gets information from the device tree to identify the available sensors and automatically provide enumerations to the end-user for using the sensors without the need for knowing the Spin pin where the sensor is connected to, or the calibration values.

### Positive changes
- This would remove a now unrequired level of hierarchy that will (slightly) accelerate user calls to Data API functions.
- Issues #29, #28 and #13 can be carried out much more simply with this change, and the work carried out when fixing these would not have to be changed in a few months (along with documentation). Indeed, without this change, fixing these issues will require changes in both APIs, work that will have to be undone when we eventually proceed to a reorganization of this kind.
- Current issue with documentation pages overlaps would be instantly fixes as a single entry point would subsist for Spin functions.
- This would introduce a new `shield` entry point which can be build upon when the Power API will be automatically generated from shield device-tree.

### Negative impacts

- There would be a slight update to the [Examples library](https://github.com/owntech-foundation/examples), where we would have to change `spin.adc.` and `data.` to `spin.data.`, as well as updating headers inclusions. A few lines should be adapted in each example.

luizvilla, 2024-05-02:
I second @jalinei on the fact that this RFC goes in the right direction. Here are my comments addressing the questions directly:  

**What is your proposed structure under shield. object ?**

For me  the resulting structure of this RFC would be :

-  Spin 
    -  data
    - comp
    - dac
    - gpio
    - led
    - pwm
    - timer
    - uart
    - version
- Shield
    - leg
    - sensors
    - filter
        - ngnd (from O2 to Twist 1.3)
        - capacitor (from Twist 1.4 onwards)
    - Safety
    - Communication
        - analog
        - can
        - rs485
        - sync
    - version

**By introducing shield. we should deprecate twist. object.**

Yes, this would replace the object indeed. 

**What would be the resulting object structure for the functions currently contained in twist. object ?
https://owntech-foundation.github.io/Documentation/core/docs/twistAPI/**

**What name do we give to the current twist. object if we place it below shield. ? legs. ? pwm. ? power. ?**

`TwistAPI` should be replaced by a `legAPI` which handles a power leg rather than the twist board, as per the structure above. 

**We do have the following issues opened on legacy gitlab forge for the lack of genericity of the current twist. object :
https://gitlab.laas.fr/owntech/power-api/core/-/issues/69
https://gitlab.laas.fr/owntech/power-api/core/-/issues/68**

The legAPI should automatically parse the dts and associate the pwms to the corresponding legs. We could associate an iteration of improving invisible mechanics of the `spin.pwm` to this process.  

**Let say that we have 2 shields, one is TWIST and the second would be OwnVerter.
How do you plan to have the right functions for the right shield using shield. approach ?**

Currently we have multiple `.dtsi` fragments associated with the twist board. I think we should build a specific `.dtsi` for each shield. This is something we already expressed as a need in the past. The `shield` object would use a similar mechanic that `dataAPI` uses to automatically setup the legs according to the `adc-channels.dtsi` content. 

**How do we deal with version. object ? we would need a counterpart of that HAL for shield. has well to define which shield will be used with a user friendly approach.**

Indeed. I've added this to my tree above.

**Can we "hide" functions from shield object if it is not relevant ?
For instance we create a third hardware shield that can only be used in buck current mode, could we make sure that shield. object only contains relevant functions ?**

We do something similar with the `data` object. We spawn different channels according to the `adc-channels.dtsi`. If we try to change the parameters of a non-existing channel, the `dataAPI` returns an error. 

I think we could spawn different legs from the shield. By revamping the `twistAPI` into a `legAPI` we could create multiple `LEG` structures which have their own setups. Some of them can become impossible if the structure does not allow it, returning an error. Others would not be available depending on an `#ifdef` structure. I think @cfoucher-laas know these codes acrobatics better than me.   

**@luizvilla do you think it still make sense to have Communication as a root structure ? Doesn't it depend of the shield as well ? Same question for Safety.**

**Does the ngnd object still make any sense under spin. ?**

I think this echoes a much more interesting aspect of our work: **modularity**. 

The communication "block" of our power boards has been reworked by @guigur as a communication shield. This means that we could have a `.dtsi` which has no power legs, but rather only the communication system. 

The measurement "block" also has its own `.dtsi` which is the `adc-channels.dtsi`. We could imagine someday having a measurement shield without the power block. Or having multiple measurement shields stacked up onto a spin board. 

The O2, Twist 1.2 and Twist 1.3 all have a `ngnd` switch which was replaced by a `capacitor` switch on the Twist 1.4. It does not exist on the OwnVerter. This kind of detail calls for a specific `.dtsi` to handle it. We could even think of a `filter` object on the `shield`. 

Finally, the `safetyAPI` is mostly shield dependent, as it depends on the voltage, power and thermal ratings of its components. So I think it should be under shield.

luizvilla, 2024-05-07:
Do we have an update on this RFC ? 
Have we achieved a conclusion on it ?

Ayoub-Farah, 2024-05-21:
I agree with the structure suggested by luiz : 

   * Spin  
     * data
     * comp
     * dac
     * gpio
     * led
     * pwm
     * timer
     * uart
     * version
   * Shield
     * leg
     * sensors
     * filter      
       * ngnd (from O2 to Twist 1.3)
       * capacitor (from Twist 1.4 onwards)
     * Safety
     * Communication       
       * analog
       * can
       * rs485
       * sync
     * version

But such an update should be kept for a V2, since this is a major structural modification and we still need to consolidate documentation. 

If we implement a shield object, then we should replace twist by leg since it is more related to power electronic. 

To differentiate twist from ownverter, we'll have to use the device tree and give specific leg configuration for each PWM. 

> Can we "hide" functions from shield object if it is not relevant ?
For instance we create a third hardware shield that can only be used in buck current mode, could we make sure that shield. object only contains relevant functions ?

It might be possible with Cmake and some Macros.

jalinei, 2024-07-15:
- Version has been suppressed in v1.0

- Spin
  -  data
    - comp
    - dac
    - gpio
    - led
    - pwm
    - timer
    - uart

-  Shield
  -  leg (contains capacitor and driver gpio)
  -  sensors
-  Safety
-  Communication
  -  analog
  -  can
  -  rs485
  -  sync

cfoucher-laas, 2024-07-19:
Moved NGND to shield. Update + small correction (leg => power) vs. above message:
- Version has been removed in v1.0
- Spin
  - comp
  - dac
  - data
  - gpio
  - led
  - pwm
  - timer
  - uart
- Shield
  - **_power_**
  - sensors
  - **_ngnd_**
- Safety
- Communication
  - analog
  - can
  - rs485
  - sync

jalinei, 2024-09-30:
test bis

jalinei, 2025-01-27:
v1.0 implement this reorganization. 
Closing

closed 2025-01-27 by jalinei