owntech-foundation/Core #47: [ADC][PWM] Improve enum name and better document Handshake for synchronous measurment.

jalinei, 2024-05-02:
Following hackathon feedback : 
It seems still hard to understand how to configure the external event that links `ADC` and `PWM`

Maybe a slight rework is necessary in the enum to change the naming and make it dumb proof ? 
- hrtim_ev1 and ADCTRIG_1 are not self explanatory. 
- hrtim is not self explanatory we should use MEAS_TRIG_1 or something like that
- Both enum should be indentical on both sides. 

In the documentation we need a dedicated paragraph to explain this handshake as it is not trivial.

cfoucher-laas, 2024-12-09:
After discussing with @jalinei:
The best user-friendly approach would be, on `Spin.data` side, to hide the number of the event. Only indicate the source of the event to be either `software` or `pwm`. This can be done by hardcoding an event number for each ADC.

Constraints are:
- We would need an event that can be ORed on multiple event sources within the HRTIM,
- We would need an event whose comparator is COMP3,
- We would need an event that can compare on any HRTIM timer unit,
- We would need that behavior to be possible for ADC 1, 2 and 3.

Data is:
- Events 1/2/3/4 can be ORed
- HRTIM can emit events 1/3 on COMP3 for timers Master/A/B/C/D/E/F
- HRTIM can emit events 2/4 on COMP3 for timers Master/E/F
- HRTIM can emit events 5/7/9 on COMP3 for timers Master/A/B/C/D/E/F
- HRTIM can emit events 5/7/9 on COMP3 for timers Master/E/F
- ADC 1/2 can receive events 1/3/5/6/7/8/9/10
- ADC 3/4/5 can receive events 1/2/3/4/5/6/7/8/9/10

We can thus select event 1 for ADC1 and event 3 for ADC2. However, there seems to be no solution that fulfill all requirements for ADC 3: either select an event that cannot be ORed, use and event that can only originate from Master/E/F, or do not use COMP3.

jalinei, 2024-12-12:
I propose the following : 

- ADC1 and ADC2 respectively behing tied to event 1 and event 3. 
- ADC3, ADC4, ADC5 behing tied to event 5, event 7 and event 9.

This way, HRTIM comp3 can be used to trigger any ADC. 
The only tradeoff in the future is that only ADC1 and ADC2 can have ORed events using multiple source aside from comp3.

jalinei, 2024-12-13:
Let's keep it simple. I can't foresee a case were we'd really need to change to different events. 
We'll still be able to add the capability in the future if it is blocking..