owntech-foundation/Core #58: [RFC] Fin a better name for the "Core" repo

cfoucher-laas, 2024-05-31:
**Is your feature request related to a problem? Please describe.**
The `Core` name was chosen at a time when there was on one side a Core API repo for OwnTech Power API and on another side the peripheral OwnTech Power API libraries (OPALIBs) repos. This situation evolved to a centralized repository made of various Zephyr modules. The name Core now seems a bit out of place.

**Describe the enhancement you'd like**
Rename the Core repository to a name better reflecting what it represents in the OwnTech ecosystem.

**Describe alternatives you've considered**
`Power API`, though it would conflicts with the Power API module of the current `Core` repository.

**Additional context**
I should maybe add that the `Core` naming was chosen by myself without any consultation, if it helps to motivate a change ;-)

jalinei, 2024-06-02:
Agree, we need to change it. 
@martinjaeger was also mentioning that. It is a bit weird to fork a the repo with default name that is "Core". 

Simple proposal -> owntech-core ? 
I'm bad at naming things, lets invoke @luizvilla on this one.

martinjaeger, 2024-06-02:
What about OwnTech Power Firmware and `owntech-power-firmware` for the repo name?