owntech-foundation/Core #68: [BOOTLOADER] Protocol error

jalinei, 2024-06-22:
**Describe the bug**
In some rare case, flashing is not working because of "Protocol error". 
It seems to be linked with `pre_bootloader_serial.py`
line 120 : 
```
	if not bool(upload_options.get("disable_flushing", False)):
		env.FlushSerialBuffer(spin_port)
```
When adding a try except to handle a failed flush serial buffer, it goes into the exception, but flashes.

**To Reproduce**
Hard to reproduce, only happens on Luiz computer so far. 

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

luizvilla, 2024-07-08:
I have not being able to reproduce the error again.

luizvilla, 2024-07-08:
The current code is this one: 
```python
def upload_pre(source, target, env):
	# Get spin port
	spin_port_ret = find_spin_port(env)
	if spin_port_ret == None:
		print("Error! No valid USB port found.")
		print("Make sure board is connected to the host computer.")
		exit(-1)

	(spin_port, spin_id) = spin_port_ret
	# Reboot board
	upload_options = {}
	if "BOARD" in env:
		upload_options = env.BoardConfig().get("upload", {})

	if not bool(upload_options.get("disable_flushing", False)):
		env.FlushSerialBuffer(spin_port)

	env.TouchSerialPort(spin_port, 1200)

```

luizvilla, 2024-07-08:
I have added the following line : 

```python
	if not bool(upload_options.get("disable_flushing", False)):
		print("Calls the fulsh of the buffer")
		env.FlushSerialBuffer(spin_port)

```
and here's the result:

```
upload_pre(["upload"], [".pio/build/USB/firmware.mcuboot.bin"])
Detecting Spin boards connected to host...
  Board 1: Found Spin board on port /dev/ttyACM0 with unique ID 463050040035003B
Uploading to board with ID 463050040035003B
Calls the fulsh of the buffer
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Rebooting board in bootloader mode...
```

luizvilla, 2024-07-08:
It does go into the flush and the buffer flushing seems to work. 

I will close this issue for now.

luizvilla, 2024-07-08:
I got the error again. 
I managed to get the error when I was testing the frequency issues. 
When initializing a very low frequency (1Hz), the micro-controller got blocked. 
When blocked I could not Flush the Serial and got the error. 
I had to put the micro-controller in recovery mode to eliminate the error.

luizvilla, 2024-07-08:
To reproduce : 

Use the PWM example and call `spin.pwm.initFrequency(1);` in the `setup_routine`.

jalinei, 2025-10-27:
@luizvilla I'll close since I think this has been solved now we introduced new initVariableFrequency and initFrequency functions.

closed 2025-10-27 by jalinei