owntech-foundation/Core #77: [NAME_OF_API] Title of your documentation issue here

pierre-haessig, 2024-07-09:
Hello,

Unless I missed something, I didn't see in the [Environment Setup](https://docs.owntech.org/core/docs/environment_setup/) doc a paragraph about how to set up the **write access permission** for the serial port (`/dev/ttyACM0`). By default I have the following file permissions on Ubuntu 22.04, where only `root` user or the `dialout` group have the write permission:

```
$ ls -l /dev/ttyACM0 
crw-rw---- 1 root dialout 166, 0 juil.  9 14:55 /dev/ttyACM0
```

In the PlatformIO doc [PlatformIO Core (CLI) / Installation / System Requirements](https://docs.platformio.org/en/latest/core/installation/requirements.html), there is a link "Please install [99-platformio-udev.rules](https://docs.platformio.org/en/latest/core/installation/udev-rules.html#platformio-udev-rules)". However, I think we cannot simply link this page because it's not adequate for OwnTech users for two reasons:

1. the provided [99-platformio-udev.rules](https://raw.githubusercontent.com/platformio/platformio-core/develop/platformio/assets/system/99-platformio-udev.rules) doesn't contain the SPIN microcontroller (with `VID:PID` identifiers printed below)
2. this doc recommends that "Ubuntu/Debian users may need to add own “username” to the “dialout” group if they are not “root”". However, this is redundant with the udev.rules. Also, the provided command also adds the user to the group  `plugdev` without explanation.

Since I don't know what's the best recommendation, I didn't submit a PR so that a consensus may be reached here before. I think there are two options to be chosen from:

1. Provide an OwnTech-specific `99-platformio-udev.rules` that gives write access to any user and document how to use this file like what is done in the PlatformIO doc I mentioned
2. Recommend instead to add the user to the appropriate group with a command like `sudo usermod -a -G dialout username`. However, this requires being sure of the appropriate group name attached to `/dev/ttyACM0`, since `dialout` is perhaps Ubuntu-specific. Also, there is the minor issue that adding a user to a group is only effective after relogging...

For the reference, I also found this doc for Arduino on Debian which is about the same problem (https://wiki.debian.org/fr/Arduino in French)

```
$ pio device list
/dev/ttyACM0
------------
Hardware ID: USB VID:PID=2FE3:0100 SER=4630500300500035 LOCATION=3-4:1.0
Description: SPIN
```

jalinei, 2024-07-09:
Thanks @pierre-haessig for pointing this out. 
This `udev` mess is indeed not documented yet. 

I guess the best option is to indeed provide a udev file that correctly adds the necessary USB ids. 

I found the following rules I've added quite a long time ago in `/etc/udev/rules.d` 

Adding the file 
`77-usb-device-owntech.rules`

```
SUBSYSTEM=="usb", ATTRS{idProduct}=="0100", ATTRS{idVendor}=="2fe3", MODE="666"
SUBSYSTEM=="usb", ATTRS{idProduct}=="0101", ATTRS{idVendor}=="2fe3", MODE="666"
SUBSYSTEM=="usb", ATTRS{idProduct}=="0102", ATTRS{idVendor}=="2fe3", MODE="666"
```
Then restarting udev using : 
```
sudo service udev restart
```
Should do the trick. 

From a documentation perspective I guess the best is to create a modified version of platformio rules where we append `SUBSYSTEM=="usb", ATTRS{idProduct}=="0100", ATTRS{idVendor}=="2fe3", MODE="666"`. 
This way we can rely on their how-to.

**Note:** the PID `0101` and `0102` where used in the past but are not used at the moment. I would keep at least `0101` as it permits to open a second virtual serial port on the same board or to reserve it if we want to change the bootloader adress to `0101` in the future.

pierre-haessig, 2024-07-10:
Thanks for sharing your udev rules. In the meantime, I've used the alternate way (adding my username to the `dialout` group).

One option would be to contribute back to the original PlatformIO udev rules at https://github.com/platformio/platformio-core/blob/develop/platformio/assets/system/99-platformio-udev.rules (if they accept such change).
I'm not familiar with the Vendor ID attribution process: is `2fe3` a registered ID of Owntech?

jalinei, 2024-07-10:
> Thanks for sharing your udev rules. In the meantime, I've used the alternate way (adding my username to the `dialout` group).
>
> One option would be to contribute back to the original PlatformIO udev [...]

I was thinking of it indeed. 2fe3 is Zephyr vendor ID. It make sense to contribute it to platformIO as they support zephyr framework. I'll submit something

jalinei, 2024-07-10:
Submitted : 
https://github.com/platformio/platformio-core/pull/4947

pierre-haessig, 2024-07-10:
Thanks. Let's wait for the feedback on https://github.com/platformio/platformio-core/pull/4947/ and then see how we proceed

pierre-haessig, 2024-07-11:
> OK, and in parallel, I've submitted PR [platformio/platformio-docs#358](https://github.com/platformio/platformio-docs/pull/358) to clarify the procedure (if we want to redirect OwnTech users to PlatformIO doc, it better be as clear as possible).

The PlatformIO documentation edit is now online: https://docs.platformio.org/en/latest/core/installation/udev-rules.html.

Now waiting for the udev rules files update https://github.com/platformio/platformio-core/pull/4947...

jalinei, 2024-07-11:
> > OK, and in parallel, I've submitted PR [platformio/platformio-docs#358](https://github.com/platformio/platformio-docs/pull/358) to clarify the procedure (if we want to redirect OwnTech users to PlatformIO doc, it better be as clear as possible).
> 
> The PlatformIO documentation edit is now online: https://docs.platformio.org/en/latest/core/installation/udev-rules.html.
> 
> Now waiting for the udev rules files update [platformio/platformio-core#4947](https://github.com/platformio/platformio-core/pull/4947)...

Great news ! I hope they will merge the udev file soon :)

pierre-haessig, 2024-07-12:
PlatformIO devs merged https://github.com/platformio/platformio-core/pull/4947

I've added the doc changes in https://github.com/owntech-foundation/Core/issues/78

jalinei, 2024-07-12:
So it will be added in next pio release.
I suggest we close the issue.

pierre-haessig, 2024-07-12:
> So it will be added in next pio release

In fact the udev rules you pushed are already in the "latest" Platformio doc: https://raw.githubusercontent.com/platformio/platformio-core/develop/platformio/assets/system/99-platformio-udev.rules. So indeed, the problem is fixed.

closed 2024-07-12 by pierre-haessig