owntech-foundation/Core #92: [platformio.ini] No support for NO_SHIELD

luizvilla, 2024-10-23:
**Describe the bug**
At this moment it is possible to choose between the ownverter and the twist boards for compiling the shields. However, there is no "no_shield" support in the code. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to platformio.ini
2. Replace board_shield_version by anything but ownverter or twist
3. Build
5. See error

**Expected behavior**
If there is no shield, the user should be able de deactivate the "shield" api

jalinei, 2024-12-11:
Reproduced on my machine today. 

There is no need for a no_shield argument. At the moment commenting out these parameters from platformio.ini does correctly suppress shield settings in the CMakeLists.txt
The culprit is to be found in KConfig..

luizvilla, 2024-12-12:
I am not sure I agree with this. How will the user only work with the spin then ? 

We have "supported shields" written somewhere. We should at least clearly state that if the user does not want a shield, the parameter should be blank.

jalinei, 2024-12-12:
_I am not sure I agree with this. How will the user only work with the spin then ?_

Just by commenting the shield and shield version lines, I agree though, comments should be improved,

I was just pointing out that the option does already exist, and the bug root cause is not PIO but somewhere in KConfig..

jalinei, 2025-06-27:
#127  Improved comments in platformio.ini, and also fixed an include issue with safety API that were causing this issue. I'm closing