owntech-foundation/Core #95: [POWER/TWIST API] Connection/Disconnect capacitor functions inverted

pierre-haessig, 2025-01-15:
Together with @jalinei, we noticed that the [TwistAPI::disconnectAllCapacitor](https://docs.owntech.org/latest/core/docs/twistAPI/#function-disconnectallcapacitor) function does the opposite of what it says, in the 0.9.0 branch. That is, to disconnect the Aluminium electrolytic capatictor, one needs to call [connectAllCapacitor](https://docs.owntech.org/latest/core/docs/twistAPI/#function-connectallcapacitor) instead!

**To Reproduce**
Steps to reproduce the behavior:
1. Load the [Voltage mode example](https://docs.owntech.org/latest/examples/TWIST/DC_DC/buck_voltage_mode/), using the 0.9.0 branch
2. Compile and upload the code
3. Check the status of the Capacitor 1 and 2 pins of the SPIN µC: I don't remember the logic behind the capacitor connection exactly, but the pin voltage is opposite of what it should be

jalinei, 2025-01-15:
MCU signal disconnect goes through the optocoupler that invert the logic:
- When GPIO is low, photodiode is off, phototransistor is opened and gate signal is high through the pull-up resistor. 
- When GPIO is high, photodiode is on, phototransistor is closed, so gate signal is pulled to ground. 
 
[image]

@luizvilla can you take care of that one ?

jalinei, 2025-06-25:
Fixed in v1.0

pierre-haessig, 2025-07-11:
So you this issue be closed?