owntech-foundation/Core #99: Clarify relationship between two scope plotting scripts

pierre-haessig, 2025-01-28:
At present (at least on v1.0.0rc branch), there are two pieces of code to plot scope results in Core (in `owntech/scripts`):

- [scripts/plot/pre_plot_records.py](https://github.com/owntech-foundation/Core/blob/v1.0.0-rc/owntech/scripts/plot/pre_plot_records.py)
- [scripts/monitor/filter_recorded_datas.py](https://github.com/owntech-foundation/Core/blob/v1.0.0-rc/owntech/scripts/monitor/filter_recorded_datas.py)

in the end, it's not clear which one is doing the actual "auto plot". From my tests, it is the `plot_df` method inside `filter_recorded_datas` which is launched automatically (along CSV file saving). Perhaps there is a way to call the other one, but I don't know how.

pierre-haessig, 2025-01-28:
A slightly related topic is to document how users should implement custom script. At present it requires changing a method in a private module (private in the sense that `owntech/scripts` is hidden in nonexpert mode).

Perhaps provide a way for people to register their own plotting function or script/filter?

jalinei, 2025-01-28:
Hi @pierre-haessig, 

- [filter_recorded_datas.py](https://github.com/owntech-foundation/Core/blob/v1.0.0-rc/owntech/scripts/monitor/filter_recorded_datas.py) This file retrieve the data from USB serial. (The stream of data is saved by adding a `monitor` filter from platformio. More info here : https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#cmd-device-monitor-filters

- [scripts/plot/pre_plot_records.py](https://github.com/owntech-foundation/Core/blob/v1.0.0-rc/owntech/scripts/plot/pre_plot_records.py) This second file register a PlatformIO action (It basically adds the input `plot recording` in platformio menu in `OwnTech` folder) it also make sure platformIO python has necessary dependencies (numpy, pandas..) The plotting is done by this file only. 

It is indeed confusing, because there is some legacy stuff in the `filter` file (we initially only had the filter file). There is some clean up to be done here. I'll try to do it in time for v1.0 release.

pierre-haessig, 2025-01-29:
>  (The stream of data is saved by adding a `monitor` filter from platformio. More info here : https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#cmd-device-monitor-filters

Reading through the doc, I didn't find if filters can take arguments (which could be used to tweak the filter, e.g. activate/disable autoplot feature for examples).
- At least, none of the default builtin filters seem to take arguments (in a sense, the filters are arguments for the monitor command...).
- But on the other hand, the filter demo suggests that the class can take arguments

```python
class Demo(DeviceMonitorFilterBase):
    NAME = "demo"

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        print("Demo filter is loaded")
...
```

jalinei, 2025-01-30:
> A slightly related topic is to document how users should implement custom script. At present it requires changing a method in a private module (private in the sense that `owntech/scripts` is hidden [...]

I'm not sure it should be accessible to begginers. In my opinion tweaking the way you interact with the automatic data plots must remain an expert feature. 

However I agree that we should create a dedicated documentation, and maybe a simple example on how to tweak the plotting parameters. 

There is a lot of additional features we could derive from the Monitor filters and the Plot scripts. We integrated what we fought being a good starting point. 

Maybe we can use this issue to create a curated list of possible improvements ? 

In the meantime, I'm marking the issue as a Documentation issue.

jalinei, 2025-06-25:
#125 Fixed

closed 2025-06-25 by jalinei