owntech-foundation/Core #113: [Upload] Issue - Setting Burst Mode PWM v1.0.0-rc

Baoxstyle, 2025-04-04:
Hello,

In the context of controlling the SPIN board, I am creating an issue ticket because I have troubles uploading the file "main.cpp" from"Setting Burst Mode PWM" (VSCode _application_ > PlatformIO _extension_ > USB _folder_ > Examples SPIN _folder_ > Setting Burst Mode PWM _file_ in the v1.0.0-rc _branch_).

To go deeper in my issue, the problem first of all is not about building of course because I didn't modify at all the "main.cpp" _file_ but it is only about uploading it in the SPIN board. Secondly, to explain how I tried to solve it here is what I did :  

- Plug/unplug the SPIN board, 
- Tried to put a space/enter in the "platformio.ini" _file_ to force re-building/compiling process just in case, 
- Close+start/restart my computer,
- Close and re-open VSCode but nothing change I can't upload to program.

I don't think it is about administrator permission because I could work with Blinky LED and Setting PWM duty cycle programs from the tutorials (Availables in the Owntech Documentation).

Finally here is my main problem : 

[image]

Thank you in advance for your precious help, wish you a nice day.

Best regards.

PIGNOL Axel

luizvilla, 2025-04-04:
This sounds strange, specially if it worked before. 
Can you verify that the Blinky still flashes on your SPIN board?

Otherwise, there is this issue from platformio that might be related. Not sure though: 
https://github.com/platformio/platformio-vscode-ide/issues/1195

Baoxstyle, 2025-04-04:
I guess, it is to change branch and upload the burst mode pwm the problem because I tried to change SPIN which worked but as soon as I tried to upload on the new SPIN board the upload fail...

Because to upload the blinky program it didn't work until I changed SPIN board... I am afraid to have broken the SPIN board by uploading the Burst Mode PWM because of it, I can't upload any problem to the past two SPIN board

luizvilla, 2025-04-04:
Can you try doing a recovery, please ? You have to press and hold the reset button then press the bootloader button and then let go the reset. The second LED will turn on and you can then flash in recovery mode.  This will allow you to flash the SPIN board again.

Baoxstyle, 2025-04-04:
I run the Bootloader recovery using USB on the board and now I can upload blinky program thank you so much, I think I will first try to understand everything before running the burst mode again. Wish you all a nice week-end and thank you so much for your precious help.

Baoxstyle, 2025-04-07:
Hello everyone, 
Hope you're all doing well, following my last issue request I wanted to mention I could build and upload "Setting Burst PWM" file but I can't watch the monitor... do you know about this error ? Have a nice day.

[image]

Baoxstyle, 2025-04-07:
Hello, it is me again, I solved the problem which was the #include "CommunicationAPI.h" because I only the SPIN board and not the Twist board. 

So I just deleted the line.

Have a nice day :)

luizvilla, 2025-04-07:
I've tried compiling and flashing the example available on the project tasks.
I've flashed the Burst Mode code. After that the serial communicaiton no longer works. 
I'm on it to find out what the problem is.

luizvilla, 2025-04-07:
We've dug a litte deeper with @jalinei and we found the problem. 

When calling the `spin.pwm.setFrequency(200000)` **before** `spin.pwm.setModulation(XXX)` the system crashes. 

This is due to the fact that `setFrequency()` does not call 

```c
	if (!hrtim_get_status(pwmX))
		/* Initialize default parameters before */
		hrtim_init_default_all();
```

Without a proper intialization, the system crashes when processing all the function calls within the `setFrequency()` function.

Baoxstyle, 2025-04-07:
Thank you so much for your precious help.

closed 2025-06-25 by jalinei