owntech-foundation/Core #120: [SPIN API] Provide access to the pwm in int

luizvilla, 2025-04-29:
**Is your feature request related to a problem? Please describe.**
When using the pwm in a more advanced setting, it is interesting to use ints instead of floats to control with more precision. 

**Describe the solution you'd like**
I would like a function call such as `setDutyRaw` which would accept ints and allow sending it straight into the lower level.

jalinei, 2025-04-29:
This is indeed needed. We've developped it at some point long ago : 

https://gitlab.laas.fr/Elie/dac-dma/-/blob/AutomatedSweep/zephyr/modules/owntech_hardware_configuration/zephyr/src/hrtim_configuration.cpp?ref_type=heads

But somehow it never made it to the main branch...

luizvilla, 2025-05-06:
I've addressed this in the latest PR #109

closed 2025-06-25 by jalinei